orangecrab-fpga/orangecrab-hardware #4: ECP5 Reset tests

gregdavill, 2019-10-16:
The ECP5 has two different resets mechanisms.

- Power on Reset
- Device Refresh

Using device refresh we can make use of the ECP5's "multi-boot" feature to load "User gateware" and "Bootloader gateware". We need to investigate if the two different reset mechanisms result in switching between these different gatewares in the same manner. 

Ideally I'd like the reset button to reset the User gateware when pressed, and then invoke the Bootloader when held down (2-3s).

If required we can make use of the Microchip ARM processor on board as a "FPGA supervisor".

et-tommythorn, 2019-12-01:
FWIW, I find it too easy to accidentally reset. I have seen lower profile/harder to press switches but I don’t know if they are small enough. Or just make the switch a user button instread.

gregdavill, 2019-12-02:
I wonder if we just make it a user accessible button... 

The loaded gateware can determine if it's been held for awhile and trigger a reset. 
Worst case power cycling the USB cable while holding it down can force a DFU mode.

gregdavill, 2019-12-08:
r0.2 design now includes this button as a user_button. So it's accessible to the gateware. 

It will be used by the bootloader to determine if the bootloader should stay active of the user gateware will run.

closed 2019-12-08 by gregdavill