orangecrab-fpga/orangecrab-hardware #6: Example gateware

gregdavill, 2019-10-16:
To get people up and running quickly some examples should be included.

- [ ] Gateware controlled RGB LED pulse/cycle.
- [ ] SoC Controlled RGB LED pulse/cycle.
- [ ] SoC with DDR3L memory mapped terminal over USB CDC.
- [ ] Bootloader

tommythorn, 2019-10-16:
FWIW, my project #0 is the simplest possible "blink LED" (optionally with button controlled speed).
Project #1 is RS232 In/Out example (typically map lower case to upper case and visa versa).

My point being that building on this is easy.  The trivial examples are valuable to pipe clean flow.  I find that most dev kits neglect the simple stuff and rush into very complicated examples with a lot of setup.

I hope to contribute simple examples :)

olofk, 2019-10-16:
I would love to see OrangeCrab in project [LED to Believe](https://github.com/fusesoc/blinky) as well

gregdavill, 2019-10-16:
I agree, but maybe my "basic" example is still to advanced. A simple blink LED is a good idea!

It may be difficult to change the speed with the onboard button, as it designed for reset only...

gregdavill, 2019-10-16:
@olofk I should be able to do that! ^_^

gregdavill, 2020-02-24:
Example projects and code are in a new repo: https://github.com/gregdavill/OrangeCrab-examples

I've already put up a few completely standalone applications for the SoC and pure verilog.

closed 2021-08-06 by gregdavill