orangecrab-fpga/orangecrab-hardware #9: Wrong value for DDR IO regulator in the CSV BOM

smunaut, 2020-01-05:
See: 
https://github.com/gregdavill/OrangeCrab/blob/01f993c3d79f494dfd4e837c9e83c070aad83b74/hardware/orangecrab_r0.1/bom/OrangeCrab.csv#L195

This should be 147k to set the regulator to 1.35v and be consistent with the schematic.

gregdavill, 2020-01-05:
Yes, 147k is the correct value.

On Commit: 215b7bc I changed the schematic and ibom to reflect this, but did not regenerate the csv file. 

Running Micron DDR3L parts at 1.5V is possible according to the datasheet, but the timing might be different. In my limited testing I could not get the standard LiteDRAM to work when the DDR3 was running at 1.5V

gregdavill, 2020-01-06:
This has been fixed in 1321707

Also updated some part numbers and swaped to the non-SerDes ECP5.

closed 2020-01-06 by gregdavill