orangecrab-fpga/orangecrab-hardware #11: Better Docs

gregdavill, 2020-02-03:
I've started working on some Docs, but these need to be improved/extended.

https://gregdavill.github.io/OrangeCrab/r0.2/

- [ ] Geting started guide, (tool-chain install to first blinky?)
- [ ] What Next (example projects/ideas)
- [ ] Step-by-Step Guide to installing Linux on the Crab.
- [ ] Get microPython/Circuit python working. (Branch from Fomu port?)

Docs are setup to use Hugo, source files are held in master, but get published to gh-pages branch. Is there a way to have them auto-generate on a new commit to master?

Anything else I'm missing?

gregdavill, 2020-07-10:
Getting Started guide now up: https://gregdavill.github.io/OrangeCrab/r0.2/docs/getting-started/

dramoz, 2021-08-13:
Hi, 

@gregdavill the link to  https://gregdavill.github.io is broken.

Regards,
Danilo.

gregdavill, 2021-08-14:
Whoops, this was a byproduct of moving the repo that I'd forgot would happen.

I'm still keen on having all the OrangeCrab repos in their own org, so I've setup a redirect from the old URL to the new one.

shenki, 2021-11-29:
While Greg works on that redirect, here's the current location of the documents:

 https://orangecrab-fpga.github.io/orangecrab-hardware/r0.2/docs/getting-started/

gregdavill, 2021-11-29:
Doh. I did have this fixed. But then I started to move my blog over to my personal github.io page.