orangecrab-fpga/orangecrab-hardware #16: [Feature Request] Chat room for help or updates

BennyTurtle, 2020-03-21:
Hello,
I honestly really love your project (But for me 100$ is kinda steep)
and I wanted to know if it would be possible to setup a chatroom (probably discord)
for all things orangecrab related. (I doubt you haven't) but if you would like help setting it up
I can help, and I'm sure there are plenty here who can also help.
Well if you do want to do this, you are free to contact me at this post or at my discord
(Bennyturtle#7936), also If you wanted example projects, once I am able to get a board I can try
to make a few basic ones to show how it works, I am a very (very) small youtuber and could make a few videos explaining it

gregdavill, 2020-03-22:
Thanks for the suggestion.
We now have a OrangeCrab channel in the 1BitSquared Discord.

https://discord.gg/c4Dnr3b
I'll add this link to the README soon.

gregdavill, 2020-07-10:
I've added this link to the README

closed 2020-07-10 by gregdavill