orangecrab-fpga/orangecrab-hardware #17: RISC-V and 128MB memory

salmansheikh, 2020-03-23:
Do you think this board/FPGA could fit a RISC-V and a memory controller to use the 128MB memory?

gregdavill, 2020-03-23:
Yep, that's exactly how I've been using it!

I've been using LiteX (https://github.com/enjoy-digital/litex) + LiteDRAM (https://github.com/enjoy-digital/litedram). 
This makes it easy to create a Vexriscv + Memory Mapped DDR3 memory controller + cache.

I'll be publishing an example of how this all works.

salmansheikh, 2020-03-24:
Cool thanks.

closed 2020-03-24 by salmansheikh

salmansheikh, 2020-03-24:
You mention 0.2 going under a short production run soon. Does that mean you will be willing to sell assembled boards? I could make pcbs but soldering those bga and other chips at home is a challenge to say the least.

gregdavill, 2020-03-24:
Ahh, yes. 

You can now join a groupbuy for boards here: https://groupgets.com/campaigns/710-orangecrab
The boards will be fully assembled by a assembly house in the US.

salmansheikh, 2020-03-25:
That's a little pricey..