orangecrab-fpga/orangecrab-hardware #18: USB-C instead of microUSB

jmi2k, 2020-03-26:
**Pros:**

* It's the new standard, and as time passes, Type C cables will be more common than micro USB
* It have better mechanical properties than micro USB
* Reversible!

**Cons:**

* It can confuse users, as the board is wired to make use of at most USB 1.1 (even High Speed is not possible according to the docs)
* microUSB cables are cheaper (though that might change).

Feather format allows for Type-C connectors to be used instead of micro USB, so it would remain Feather-compatible.

rowanG077, 2020-03-26:
USB 3.0 is possible using the ECP5-5G variant.

gregdavill, 2020-03-26:
It's a good idea. Swapping in a USB2.0 only USBC connector would be a quite easy change, and not effect any other hardware one the board.

While USB3.0 is technically possible with the SERDES that the ECP5/ECP5-5G has (3.2Gbits/5Gbits) I've not seen this working yet. So I'm hesitant to rush to put that in a r0.3 hardware version. 
That said @enjoy-digital is making good progress on that front here: https://github.com/enjoy-digital/usb3_pipe

Routing SERDES into the USB3 lines does not automatically guarantee we can operate in alternate modes (DisplayPort/HDMI) that's another can of worms using the USB billboard system, and might mean we need to include a dedicated USBC PD controller.

jmi2k, 2020-03-26:
I was more concerned about the connector itself more than the protocol. It's awesome to see that you like the idea of putting an Type-C connector while keeping the underlying USB 2.0 system, and I would love to receive my board with it.

About the protocol, I prefer a simple and clean board design (as there are alternative boards which already offer more capabilities, including USB 3.0). In fact, I was astonished when I saw the schematics because there is little hidden: it's a FPGA attached to some DRAM and little else. It's a really understandable system. However I can understand the appeal of a full-featured USB-C connector...

gregdavill, 2020-03-26:
I am open to changing it. Unfortunately, at-least at the time of writing this, we already have the design locked in for the group-buy. So it's not possible to alter those designs as they're already in the pipeline.

So this change would likely come on a r0.3 board release.

jmi2k, 2020-03-26:
It's fine, I hope there is a group-buy for r0.3 too

gregdavill, 2020-09-21:
Implemented in r0.2.1: https://github.com/gregdavill/OrangeCrab/issues/25

closed 2020-09-21 by gregdavill

ddiss, 2024-09-01:
> While USB3.0 is technically possible with the SERDES that the ECP5/ECP5-5G has (3.2Gbits/5Gbits) I've not seen this working yet. So I'm hesitant to rush to put that in a r0.3 hardware version. That [...]

Not sure whether this is still of interest, but @ktemkin proposed an alternative USB3 LFPS implementation using the ECP5 SerDes:
https://lab.ktemkin.com/post/serdes-lfps/