orangecrab-fpga/orangecrab-hardware #20: PWM control of LED in Linux

pdp7, 2020-03-28:
Please tell me if I should create this issue somewhere else.

I remember you controlled the LED with PWM driver in Linux.

Could you share the commands you used?

thanks!

gregdavill, 2020-03-29:
You need to add the `rgb-led` to the board capabilities, on this line: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L223

Then the 3 PWM drivers should be in the device tree, and accessed using these commands:
https://github.com/litex-hub/linux-on-litex-vexriscv/issues/44#issuecomment-539389442

pdp7, 2020-03-29:
@gregdavill does this mean I will have to load everything over UART?

Or is there a branch with your flash changes that I can use?

I have not been able to get SD card boot to work yet (I've tried different cards but no luck getting past Liftoff).

pdp7, 2020-03-29:
Not sure if this is the right place to post...
but it seems to stale at 0%
```
pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ lxterm --images=images.json /dev/ttyACM0 --speed=1e6 --no-crc 
[LXTERM] Starting....

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 30 2020 00:40:08
 BIOS CRC passed (5c9339f0)

 Migen git sha1: --------
 LiteX git sha1: --------

--=============== SoC ==================--
CPU:       VexRiscv @ 48MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  131072KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11100000| delays: 01+-01
best: m0, b0 delays: 01+-01
m1, b0: |11100000| delays: 01+-01
best: m1, b0 delays: 01+-01
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 82Mbps Reads: 131Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0x40000000 (5082532 bytes)...
|>                    | 0%

```

gregdavill, 2020-04-01:
This branch should work.
https://github.com/gregdavill/linux-on-litex-vexriscv/tree/orangecrab

Right now it's just setup for serial boot, but good to confirm that works first before adding FLASH back in. This should have a PWM enabled RGB led driver available in Linux.

pdp7, 2020-04-02:
@gregdavill thanks, I was able to load images.json with lxterm and boot Linux.

I ran this:
```
root@buildroot:~# cd /sys/class/pwm/pwmchip0
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0# ls
device     export     npwm       subsystem  uevent     unexport
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0# echo 0 > exp
ort
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0# cd pwm0/
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# ls
capture     duty_cycle  enable      period      polarity    uevent
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# echo 10
0 > period
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# echo 50
 > duty_cycle
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# echo 1 
> enable
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# echo 75
 > duty_cycle
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# echo 25
 > duty_cycle
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# echo 0 
> enable
root@buildroot:/sys/devices/platform/soc/f0004000.pwm/pwm/pwmchip0/pwm0# 
```

but I see no change in the LEDs on board.  I am using a buildroot that I built using the linux-on-litex-vexriscv instructions.  This is the kernel:
```
Linux buildroot 5.0.13 #1 Fri Feb 14 22:37:08 CET 2020 riscv32 GNU/Linux
```
any suggestions?