orangecrab-fpga/orangecrab-hardware #24: UDQS/LDQS nets are switched on the SDRAM

benvanik, 2020-05-26:
I noticed these two differential signals were swapped in the r0.2 schematic. I've never done a DDR design and was wondering if there was a reason for this or if it snuck in during layout?

[image]

gregdavill, 2020-05-26:
Yes, they were switched to improve routing, You'll notice that Upper Data (UDM, UDQS) is actually paired with D0-D7 bits on the FPGA end.

Ideally these nets should be renamed, to match what they're actually connected to.

krystian-hebel, 2023-11-16:
DQSs are connected properly (although signal name is wrong), but DMs are not. As a result, when bytes are written one at a time and read back in group of 4 bytes, they are swapped. We discovered this in [this function](https://github.com/Dasharo/neorv32/blob/6e0197ea210247cbffa77e3d9da7f2c56fd36899/sw/bootloader/bootloader.c#L681), with debug lines added - `recv` is the data from `PRINT_GETC()` and `signature` is what the function returns:

```
Awaiting neorv32_exe.bin...
recv: 0x000000fe
recv: 0x000000ca
recv: 0x00000088
recv: 0x00000047
signature: 0x8847feca

ERR_EXE
```

The problem is somewhat hidden with cache enabled, because all accesses are at least 4 bytes long in that case.