orangecrab-fpga/orangecrab-hardware #27: SPI flash bootloader protection

DurandA, 2020-07-14:
Hi @gregdavill,

I just got my hands on OrangeCrab and am looking to use the internal SPI flash within a LiteX SoC to store the BIOS + firmware.

I was wondering if the r0.2 as sold by GroupGets has the [W25Q128JVPIQ SPI flash](https://www.winbond.com/resource-files/w25q128jv%20spi%20revc%2011162016.pdf) preconfigured with block/sector lock in order to protect the first 4Mbits of the bootloader.

gregdavill, 2020-07-14:
No, the flash hasn't been locked. That is probably something we should have done, but totally slipped my mind when setting up the production test suite.

When I update the boot-loader. Which we can setup as a .dfu update, I think it will likely unlock/lock those bits. To prevent accidental overwriting of that area.

DurandA, 2020-07-15:
Thanks for the answer. Feel free to close this issue or keep it open until you update the bootloader.

Btw, the board is awesome!