orangecrab-fpga/orangecrab-hardware #33: OrangeCrab r0.2 DDR3 gets hot even if .pcf does not use it

tambewilliam, 2021-01-06:
Please find attached zipped .dfu file.

Below is the content of the .pcf file that was used to build the attached zipped .dfu file.

LOCATE COMP "clk48mhz_i" SITE "A9"; IOBUF PORT "clk" IO_TYPE=LVCMOS33; FREQUENCY PORT "clk48mhz" 48 MHZ;

FREQUENCY NET "clk24mhz" 24 MHZ;
FREQUENCY NET "clk96mhz" 96 MHZ;

LOCATE COMP "usr_btn_n" SITE "J17"; IOBUF PORT "usr_btn_n" IO_TYPE=SSTL135_I;

LOCATE COMP "led_red_n"   SITE "K4"; IOBUF PORT "led_red_n"   IO_TYPE=LVCMOS33;
LOCATE COMP "led_green_n" SITE "M3"; IOBUF PORT "led_green_n" IO_TYPE=LVCMOS33;
LOCATE COMP "led_blue_n"  SITE "J3"; IOBUF PORT "led_blue_n"  IO_TYPE=LVCMOS33;

LOCATE COMP "usb_d_p" SITE "N1"; IOBUF PORT "usb_d_p" IO_TYPE=LVCMOS33;
LOCATE COMP "usb_d_n" SITE "M2"; IOBUF PORT "usb_d_n" IO_TYPE=LVCMOS33;
LOCATE COMP "usb_pullup" SITE "N2"; IOBUF PORT "usb_d_n" IO_TYPE=LVCMOS33;

LOCATE COMP "sdcard_clk" SITE "K1"; IOBUF PORT "sdcard_clk" IO_TYPE=LVCMOS33;
LOCATE COMP "sdcard_di" SITE "K2"; IOBUF PORT "sdcard_di" IO_TYPE=LVCMOS33;
LOCATE COMP "sdcard_do" SITE "J1"; IOBUF PORT "sdcard_do" IO_TYPE=LVCMOS33;
LOCATE COMP "sdcard_dat1" SITE "K3"; IOBUF PORT "sdcard_dat1" IO_TYPE=LVCMOS33;
LOCATE COMP "sdcard_dat2" SITE "L3"; IOBUF PORT "sdcard_dat2" IO_TYPE=LVCMOS33;
LOCATE COMP "sdcard_cs_n" SITE "M1"; IOBUF PORT "sdcard_cs_n" IO_TYPE=LVCMOS33;

[orangecrab02.zip](https://github.com/gregdavill/OrangeCrab/files/5773583/orangecrab02.zip)

gregdavill, 2021-01-22:
I tested this code on some boards here and didn't notice any significant heat from the DDR3.

Maybe a slight 5C increase, but that's likely heat from the FPGA itself spreading into the board.

tambewilliam, 2021-01-23:
Perhaps my board has failed; because the heat is definitely coming from the DDR3 spreading quickly on PCB and can be felt while holding the board by the edges.

Would you be interested in looking at it if I shipped it to you ?

gregdavill, 2021-01-24:
Yeah, I'd be interested to take a look. 

Send me an email `greg.davill@gmail.com` 
I can ship you a new one.

gregdavill, 2021-03-26:
Just to wrap up this issue, here is a summary

I received the 'failed' board. It appeared that VCC_DRAM was ~2.8V. This should be 1.35V for DDR3L (or 1.5V for DDR3). It's supplied by one of the DCDC regulators.

Upon inspection of the regulator under a microscope a small crack extending through the WLCSP die was observed. This is presumed to be the culprit. of the fault.

A higher VCC_DRAM will not adversely effect the ECP5, as it's IO pins are all rated to 3.3V. However it was stressing the DDR3 part, leading to the DDR3 part becoming hot while powered, even if not being used.

[image]
[image]

closed 2021-03-26 by gregdavill