orangecrab-fpga/orangecrab-hardware #35: RISC-V with custom gateware

fdarling, 2021-03-28:
Are there any examples of programming an OrangeCrab using the soft RISC-V core, but with custom "gateware" for hardware accelerated peripherals? My goal is to largely program using C/assembly, but to also extend the CPU core using Verilog.

In the RISC-V blinky example, I am confused as to where the RISC-V core actually resides. Is it part of the bootloader, which simply runs the (assembly) program stored in flash rather than reconfiguring the FPGA? Or is there a RISC-V core being pulled into the blinky example that becomes part of "custom" FPGA configuration which then runs the assembly code?

I'd appreciate it if you'd shed some light on this!

tommythorn, 2021-03-29:
You don't say which blinky example (there are many), but assuming it's https://github.com/gregdavill/OrangeCrab-examples/tree/main/riscv/blink then you are right, it's executed by the RISC-V core that's part of the initial FPGA configuration.  However for example https://github.com/gregdavill/OrangeCrab-examples/tree/main/verilog/blink is a standalone logic design that would override the initial design.

If you want to extend the firmware core, then I *think* the sources are in https://github.com/gregdavill/OrangeCrab-test-sw (it's actually not clear as it appears to be a test fixture and I haven't tried it).

boxfire, 2021-09-10:
I started from the bootloader to do this: https://github.com/gregdavill/foboot/tree/OrangeCrab

boxfire, 2021-09-10:
And also: https://github.com/gregdavill/linux-on-litex-vexriscv/tree/orangecrab

francis2tm, 2022-01-17:
Hey @fdarling, I'm exactly in your situation. I pretend to add a custom peripheral into a SoC with a RISC-V core. It seems orangecrab's repos are not centrailized. So far, I've found the CircuitPython SoC (https://github.com/orangecrab-fpga/orangecrab-examples), but it seems that it doesn't have support for common peripherals like SPI, I2C etc. Whereas the SoC from https://github.com/gregdavill/OrangeCrab-test-sw/tree/main/hw looks like it is more complete. However, it is hard to know which one is more complete/updated and if they both run at maximum frequency (which is 48MHz right?) in order to use it as a baseline. Which orangecrab SoC should I start on? @gregdavill  any thoughts?

Thanks in advance