orangecrab-fpga/orangecrab-hardware #39: Documentation for r0.2.1

nicolasnoble, 2021-07-22:
I got a pair of boards from https://1bitsquared.com/products/orangecrab and what I received are in fact r0.2.1 boards, which I can't find documentation for. When connecting the boards through USB, they start the default blinking bitstream, but no usb device appears in my kernel logs, even though I'm using a data USB-C cable. I'm not too sure how to proceed without documentation about them to troubleshoot what's going on.

gregdavill, 2021-07-23:
Hey @nicolasnoble, The docs are identical to the r0.2 board, except for a connector change for the USB/JTAG, and some changes to the DC/DC circuits.

If you hold down the button while connecting the USB then the DFU bootloader will appear.

dramoz, 2021-08-14:
@gregdavill I think you should acknowledge this in the [README.md](https://github.com/orangecrab-fpga/orangecrab-hardware/blob/d53c08cea15ff8d6b18bcf5d190d7da70abeee98/README.md) file and the [OrangeCrab Web site](https://orangecrab-fpga.github.io/orangecrab-hardware/) to avoid confusions. I have to figure out the same issue.