orangecrab-fpga/orangecrab-hardware #44: I can add your board to IceStudio IDE

benitoss, 2021-10-04:
Hello I am a FPGA developer

If you want to add your board to IceStudio you only have to send me one board, and I am glad to include the board in the supported list of boards of this IDE.

IceStudio is a visual editor for open FPGA boards
https://icestudio.io/

I have already added different boards with ECP5 Lattice Model to IceStudio
Colorlight boards --> FPGAwars/icestudio#500
FleaFPGA-Ohm ECP5 board --> FPGAwars/icestudio#521
and ECP-5 Evaluation Board --> FPGAwars/icestudio#541

My email is benitoss@gmail,com if you are interested

Rergards
Fernando

gregdavill, 2021-10-05:
Hey Fernando,

Thanks for your suggestion.
To my knowledge the OrangeCrab board has already been added to ice-studio

https://github.com/FPGAwars/icestudio/blob/b13f850a2e5c594df14a0c72538ce73a7445d771/app/resources/boards/menu.json#L55

benitoss, 2021-10-06:
Ohhh yes, You are right. Sorry for the confusion.
Regards

closed 2021-10-06 by benitoss