orangecrab-fpga/orangecrab-hardware #52: USB Data signals (USB_D+, USB_D-) are not connected to a pin pair on ECP5

francis2tm, 2022-06-04:
Hey!
I was checking ECP5's [datasheet](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiOo4fzxZL4AhWmh_0HHczUBWwQFnoECB8QAQ&url=https%3A%2F%2Fwww.latticesemi.com%2F-%2Fmedia%2FLatticeSemi%2FDocuments%2FDataSheets%2FECP5%2FFPGA-DS-02012-1-9-ECP5-ECP5G-Family-Data-Sheet.ashx%3Fdocument_id%3D50461&usg=AOvVaw33dHhNVnbhpn4szaYe1r9i) and on 4.1 Signal Descriptions it says: "

> PIO A and B are grouped as a pair, and PIO C and D are group as a pair. Each pair
supports true LVDS differential input buffer. Only PIO A and B pair supports true
LVDS differential output buffer.

I wonder why the USB data signals (differential) are not connected to a pin pair (for instance, to PR44C and PR44D)?

Thanks in advance

gregdavill, 2022-06-04:
Yes you're right differential output buffers are only on PIO A/B pairs. But this doesn't matter, and isn't actually possible for Full-Speed USB signals for two reasons.
 - Against popular belief, Full Speed USB is not always a differential signal. EOP symbol is low on both D+/D-. So even if USB signals were on an A/B pair you would need to use pins in single ended mode.
 - Even without the differential buffer in the PIO, adding an inverter in the FPGA fabric works,but potentially adds some skew to the signals. This would only be an issue with signals in the 200-400Mbit/s range. USB FS being 12Mbit/s this works perfectly.

Here is a photo of the low level USB signalling which shows these EOP.
[image]

francis2tm, 2022-06-04:
Makes sense! Thanks for the clarification

closed 2022-06-04 by francis2tm