orangecrab-fpga/orangecrab-hardware #53: LiteDRAM's ECP5 IOs

francis2tm, 2022-08-06:
Hello,
I was checking your FPGA-DDR connections and it raised me 2 questions:

1. I can see that there're multiple pads that are used by LiteDRAM as "VCCIO", these pins are connected to P1.35V. The same happens to 2 GND pads. What's really the purpose of these pins? Why are these VCCIO pins connected to K16 D17 K15 K17 B18 C6 specifically? Could the VCCIO (and the GND pins) be connected to other pads of the FPGA?
2. Besides all the differential pairs that need to be connected to a "A" and "B" or a "C" and "D" pins of the FPGA, is there any constraint to consider when choosing which pads of the FPGA will be attributed to each DDR signal?
Thanks in advance!

gregdavill, 2022-08-07:
Lattice recommends assigning spare I/Os in each bank to "Virtual I/O power". Basically configuring an I/O as an output, and tying it to 1.35V results in an additional BGA ball used to power the other switching I/Os on that bank, same is true for GND. I never ran any simulations based on the speed of the LiteDRAM controller to determine if these pins actually are required for a functional design.

Yes. Checkout Lattices app notes for DDR interfaces with the ECP5. There are a few constrains around the DQS groups, the strobes are on fixed pins, and only a handful of I/O around them can be paired with that strobe. Address/Command signals don't have any restrictions.

francis2tm, 2022-08-09:
Thanks for the useful pointers!

closed 2022-08-09 by francis2tm