orangecrab-fpga/orangecrab-hardware #56: Error compiling Verilog examples with Yosys 0.37 on Mac OS

frantic0, 2024-01-28:
Hello, 

I have been trying to compile and upload with the latest [Yosys 0.37+34 oss-cad-suite-darwin-x64-20240127.tgz](https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2024-01-27/oss-cad-suite-darwin-x64-20240127.tgz) on Mac OS Sonoma 14.2

I can successfully compile and upload the blink verilog examples but not the pll, pwm_rainbow, usb_acm_device examples, which are of interest. 

I am adding the result of the make on pll. Any ideas of what is not working?

Thank you

```
$ make
ecppll -i 48 -o 100 -f pll.v
Pll parameters:
Refclk divisor: 12
Feedback divisor: 25
clkout0 divisor: 6
clkout0 frequency: 100 MHz
VCO frequency: 600
yosys -s "plltest.ys"

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.37+34 (git sha1 80511ced7, x86_64-apple-darwin21.4-clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

-- Executing script file `plltest.ys' --
ERROR: Can't open script file `plltest.ys' for reading: No such file or directory
make: *** [plltest.json] Error 1
```

`

jrsa, 2024-04-28:
install gnu make with `brew install make` and then build using `gmake` instead of `make`

frantic0, 2024-04-28:
Thank you @jrsa 

not sure why, but the makefile entry for generating the .ys files wasn't working for me. 

I was able to build it with `make` by changing the makefile and replacing the following 
```
%.ys: $(VERILOG_FILES)
 	$(file >$@)
 	$(foreach V,$(VERILOG_FILES),$(file >>$@,read_verilog $V))
 	$(file >>$@,synth_ecp5 -top $(TOP_MODULE)) \
 	$(file >>$@,write_json "$(basename $@).json") \
```
with this
```
%.ys: $(VERILOG_FILES)
	@echo "Generating Yosys script $@"
	@echo "read_verilog $(VERILOG_FILES)" > $@
	@echo "synth_ecp5 -top $(TOP_MODULE)" >> $@
	@echo "write_json $(basename $@).json" >> $@
```

jrsa, 2024-04-28:
it's most likely because something in the original makefile is only supported by GNU make, and not by BSD make which is what ships with OS X