orangecrab-fpga/orangecrab-hardware #57: Restoration of DFU bootloader

jrsa, 2024-04-28:
I made the regrettable mistake of overwriting the DFU bootloader by leaving out the "offset" parameter when invoking `dfu-util` and I am trying to restore it via JTAG with an FT2232H breakout board, and the `ecpprog` tool. I have the 25F variant, revision 0.2.1.

I tried flashing the bootloader bitstream from [here](https://github.com/gregdavill/foboot/releases/tag/v3.0) and get the following output:
```
ecpprog -I B  foboot-v3.0-orangecrab-r0.2-25F.bit  -a
init..
IDCODE: 0x41111043 (LFE5U-25)
ECP5 Status Register: 0x00200100
reset..
flash ID: 0xEF 0x40 0x18
file size: 266307
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
erase 64kB sector at 0x020000..
erase 64kB sector at 0x030000..
erase 64kB sector at 0x040000..
programming..  266307/266307
verify..       266307/266307  VERIFY OK
rebooting ECP5...
Bye.
```
That seems... good. However after the device resets there is a red breathing pattern on the RGB LED and no DFU device shows up. I am assuming the red LED indicates an error with the bootloader. The result is the same whether I plug in the device with the button pressed or not. I am successfully able to load the blink example in this manner so I think my process for uploading a bitstream to flash is OK. My only other idea is that I am using the wrong bitstream for the hardware I have, but I don't see any on github that might be a closer match.

jrsa, 2024-04-29:
I also tried this bitstream with the same result
[https://github.com/orangecrab-fpga/production-test-sw/blob/main/prebuilt/foboot-v3.1-orangecrab-r0.2-25F.bit](https://github.com/orangecrab-fpga/production-test-sw/blob/main/prebuilt/foboot-v3.1-orangecrab-r0.2-25F.bit)

gregdavill, 2024-05-07:
That should be all that's needed to load the bootloader via JTAG.

Are you holding the button when connecting USB? That's needed to "stay" in the bootloader, otherwise it tries to run the user application.

jrsa, 2024-05-12:
yes @gregdavill I am pressing `btn0` and then inserting the USB-C. The RGB LED blinks red immediately after the cable is inserted. I looked through what I believe to be the bootloader source [here](https://github.com/gregdavill/foboot/blob/4f24c432a84b1f2ddadfc2001b087afb6176b0dc/sw/src/main.c#L49) to see why the red blinking occurs (I am assuming that I see the results of `rgb_mode_error`)
I see several calls once inside the dfu handling code, but I am assuming that is not getting reached because it never enumerates on USB. I believe USB is physically working because I am able to load and run the CDC ACM verilog example. The line I linked above is the only other `rgb_mode_error` which seems to indicate failing to reboot to the CPU (the LED state is set right before doing so). Is it possible the firmware is trying to reboot from flash memory, and failing, because I haven't written any firmware to flash? Not entirely sure how this is supposed to work, but I was expecting if this bitstream was loaded, that would be enough to do DFU bitstream uploading.

CCampbell71, 2024-12-14:
@jrsa did you figure this out? I'm planning on building a board from scratch and want to ensure i can get the bootloader functioning

jrsa, 2024-12-14:
@CCampbell71 I gave up since I have other ECP5 dev boards, but I'm guessing I bricked mine somehow while trying to fix this issue (which of course originated with _my_ mistake) Good luck with the build!

CCampbell71, 2024-12-15:
@jrsa oh no! @gregdavill on a freshly assembled board, I should be able to flash the bootloader via JTAG? The prebuilt one in your repo?

gregdavill, 2024-12-15:
It should be all that is needed. That is what's loaded onto the boards during their assembly QC. The JTAG interface is part of the FPGA and is always available to load gateware,

The bootloader project should still run and compile too, if you needed to change the behaviour around the btn0 input.